eclipse-archived/kiso #24: Move docker image to Eclipse Foundation public registry

HansUweRempler, 2020-01-17:
[image] **Issue by [bierkandt-bosch](https://github.com/bierkandt-bosch)**
_Monday Sep 23, 2019 at 08:36 GMT_
_Originally opened as https://github.com/Bosch-AE-SW/cddk-oss/issues/268_

----

Hide link to Bosch internal Docker registry => move docker image to public registry

HansUweRempler, 2020-01-17:
[image] **Comment by [vkonov](https://github.com/vkonov)**
_Wednesday Sep 25, 2019 at 07:39 GMT_

----

According to [DockerHub Pricing](https://hub.docker.com/pricing) for organizations there is no free tier.
We could setup a free individual account to hold the repository with the image, but we would need to share the account (of course bad idea) and it probably clashes with docker hub's usage policies.

@bierkandt-bosch, do you know who we can contact to see if there is already a Bosch organization in docker hub, how we can get a new team in it, if there is one, and make a repository?

HansUweRempler, 2020-01-17:
[image] **Comment by [bierkandt-bosch](https://github.com/bierkandt-bosch)**
_Wednesday Sep 25, 2019 at 11:26 GMT_

----

We could work only with docker file, can be difficult in company networs
- Does EF using docker?
  - Check with Jens Reimann
- Does Bosch use docker?
  - Check with SW Campus Tools Team
  - Check with eBike  OE Dashboard Team
  - Check with eBike BDP Team
  - Check with Steffen Evers (INST)

HansUweRempler, 2020-01-17:
[image] **Comment by [sebastianpfischer](https://github.com/sebastianpfischer)**
_Wednesday Sep 25, 2019 at 11:29 GMT_

----

> 
> 
> We could work only with docker file, can be difficult in company networs
> 
>     * Does EF using docker?
>       
>       * Check with Jens Reimann
> 
>     * Does Bosch use docker?
yes
>       
>       * Check with SW Campus Tools Team
yes -> intern registory
>       * Check with eBike  OE Dashboard Team
yes
>       * Check with eBike BDP Team
>       * Check with Steffen Evers (INST)
yes, they have a linux VM running with what they call SAZ connection to internet.

--> but still check :)

HansUweRempler, 2020-01-17:
[image] **Comment by [yongbosch](https://github.com/yongbosch)**
_Wednesday Sep 25, 2019 at 12:44 GMT_

----

Does EF using docker?
https://www.eclipse.org/projects/handbook/#resources-dockerhub

This how E.F. use docker

HansUweRempler, 2020-02-14:
We believe this is done by using dockerhub, see
https://github.com/eclipse/kiso/blob/master/Jenkinsfile

---
        kubernetes
        {
            containerTemplate
            {
                name 'kiso-build-env'
                image 'eclipse/kiso-build-env:v0.0.1'
                ttyEnabled true
                resourceRequestCpu '2'
                resourceLimitCpu '2'
                resourceRequestMemory '8Gi'
                resourceLimitMemory '8Gi'
            }
        }

closed 2020-02-14 by HansUweRempler

sebastianpfischer, 2020-02-14:
This is mostly to tell to use the docker eclipse/Kiso-env image.