eclipse-archived/kiso #31: Enable Kiso to be build as 3rd party library

HansUweRempler, 2020-01-17:
[image] **Issue by [sebastianpfischer](https://github.com/sebastianpfischer)**
_Tuesday Oct 08, 2019 at 09:46 GMT_
_Originally opened as https://github.com/Bosch-AE-SW/cddk-oss/issues/307_

----

Projects using Kiso would probably have their own build system. Nevertheless, building an application on Kiso does not mean I need to integrate my application to Kiso.
Therefore the following parameters need to be added:
* KISO_APPLICATION_PATH allowing a project to link an external application to build with Kiso
* KISO_CONFIGURATION_PATH allowing a project to link an external configuration to use to build the Kiso packages/libs
* KISO_BOARD_PATH allowing a project to link an external board to use to build the Kiso HAL

As default:
* KISO_APPLICATION_PATH set to examples/${KISO_BOARD_NAME}/
* KISO_BOARD_PATH set to boards/${KISO_BOARD_NAME}/
* KISO_CONFIGURATION_PATH boards/${KISO_BOARD_NAME}/config

HansUweRempler, 2020-01-17:
[image] **Comment by [sebastianpfischer](https://github.com/sebastianpfischer)**
_Thursday Oct 10, 2019 at 08:20 GMT_

----

Discussion:
* Define UCs
* Analyse them
* Implement a solution
--> needed for ICD

HansUweRempler, 2020-01-17:
[image] **Comment by [bierkandt-bosch](https://github.com/bierkandt-bosch)**
_Tuesday Oct 15, 2019 at 09:54 GMT_

----

Milestone planning:
- Approach:
  - Define use cases
   - Prioritize them
   - Analyse them
  - have an alignment and decision on configurability (what to do for ICD, what to do after)
- Other related issues:
- #297 
- #302

HansUweRempler, 2020-01-17:
[image] **Comment by [elbosch](https://github.com/elbosch)**
_Thursday Nov 28, 2019 at 09:30 GMT_

----

The ticket title and the description are a bit misleading. Here we are talking about building applications and BSPs from different locations then the Kiso examples folder. 

If we want to treat Kiso as 3rd party library we would need to have build target for kiso which would build everything except application and BSP, put it in a 3rd party lib -> have a libkiso.a afterwards

How would someone add its own third party libs to the project, that are not in the Kiso 3rd Party folder? Shall he add those in his fork or shall we provide also a mechanism for the user to specify his own library folder?

HansUweRempler, 2020-01-17:
[image] **Comment by [vkonov](https://github.com/vkonov)**
_Thursday Dec 12, 2019 at 08:59 GMT_

----

Additional changes from meeting and PR #297 comments:
- [ ] Separate BSP interfaces in their own library
- [ ] Remove ENABLE_ALL_FEATURES flag in favor of using the new config directory approach
- [ ] Restructure config directory to reflect actual library structure in Kiso
- [ ] Enable usage of board-specific configuration headers
- [ ] Check cmake files structure

vkonov, 2020-02-14:
Done by https://github.com/eclipse/kiso/pull/90

closed 2020-02-14 by HansUweRempler