eclipse-archived/kiso #33: Document Cellular

HansUweRempler, 2020-01-17:
[image] **Issue by [sebastianpfischer](https://github.com/sebastianpfischer)**
_Tuesday Oct 08, 2019 at 10:26 GMT_
_Originally opened as https://github.com/Bosch-AE-SW/cddk-oss/issues/320_

----

*Simplicity*
- [x] Add within the website the concept --> Hugo
- [ ] Add code snipped --> after Hugo

Please note that this issue is not about documenting the cellular example "CommonGateway/c-cellular/" but the SW module documentation, different issue -> #330

HansUweRempler, 2020-01-17:
[image] **Comment by [HansUweRempler](https://github.com/HansUweRempler)**
_Wednesday Dec 04, 2019 at 05:44 GMT_

----

I assume the corresponding branch is https://github.com/Bosch-AE-SW/cddk-oss/tree/feature/%23193-document-cellular, right? If this is the case, please add the branch to the issue description.

HansUweRempler, 2020-01-17:
[image] **Comment by [HansUweRempler](https://github.com/HansUweRempler)**
_Friday Dec 13, 2019 at 08:51 GMT_

----

Code snippet not rendered in Hugo -> we didn't find it today

HansUweRempler, 2020-02-14:
Code snippet not yet showing in Doxygen: http://kiso.rempler.de:8080/group___c_e_l_l_u_l_a_r.html

ChiefGokhlayehBosch, 2020-02-14:
I have a commit (as part of [prototype/attransceiver](https://github.com/ChiefGokhlayehBosch/kiso/tree/prototype/attransceiver) in my personal [fork](https://github.com/ChiefGokhlayehBosch/kiso)) cleaning up several doxygen warnings and fixing files not showing up. Feel free to cherry-pick.

https://github.com/ChiefGokhlayehBosch/kiso/commit/04c73611dd1d421ceb93a9c71d14a7139b0e370f

It's not perfect, but will be updated while working on [prototype/attransceiver](https://github.com/ChiefGokhlayehBosch/kiso/tree/prototype/attransceiver).

See you soon! :)

ChiefGokhlayehBosch, 2020-02-21:
* [x] Add code snippets to doxygen documentation for easier driver integration
* [x] Document variant management concept (in doxygen)
* [x] Correct and expand Cellular Developer's Guide on hugo

elbosch, 2020-02-21:
solved in #131

closed 2020-02-21 by elbosch