eclipse-archived/kiso #161: STM32F7 default build not working

HansUweRempler, 2020-03-13:
Bug:

```
rempler@RT-Z0ITK:~/r/eclipse$ git clone https://github.com/eclipse/kiso.git
Cloning into 'kiso'...
remote: Enumerating objects: 3282, done.
remote: Total 3282 (delta 0), reused 0 (delta 0), pack-reused 3282
Receiving objects: 100% (3282/3282), 23.09 MiB | 1.79 MiB/s, done.
Resolving deltas: 100% (1212/1212), done.
Checking out files: 100% (2202/2202), done.
rempler@RT-Z0ITK:~/r/eclipse$ cd kiso
rempler@RT-Z0ITK:~/r/e/kiso:master$ cmake . -Bbuild -DKISO_BOARD_NAME=NucleoF767
-- Looking for sysroot: /usr/lib/arm-none-eabi
------------- KISO CONFIG -------------
Building Kiso tests:   OFF
  ... with coverage:   ON
Kiso Board Path:       boards/NucleoF767
Kiso OS:               freertos
Kiso Application Path: examples/NucleoF767/c-leds
Project Config Path:   /home/rempler/repos/eclipse/kiso/config
------------- KISO CONFIG -------------
-- The C compiler identification is GNU 6.3.1
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- JLink not found in PATH. "flash" target unavailable!
ERROR JLink not found in path!
/bin/wslpath: /home/rempler/repos/eclipse/kiso/build/applications/c-leds/application.hex: Invalid argument
CMake Error at cmake/FlashTarget.cmake:65 (string):
  string sub-command STRIP requires two arguments.
Call Stack (most recent call first):
  examples/NucleoF767/c-leds/CMakeLists.txt:22 (CREATE_FLASH_TARGET_JLINK)

/bin/wslpath: /home/rempler/repos/eclipse/kiso/build/applications/c-leds/flash.jlink: Invalid argument
CMake Error at cmake/FlashTarget.cmake:83 (string):
  string sub-command STRIP requires two arguments.
Call Stack (most recent call first):
  examples/NucleoF767/c-leds/CMakeLists.txt:22 (CREATE_FLASH_TARGET_JLINK)

-- Applying board-specific configuration from /home/rempler/repos/eclipse/kiso/boards/NucleoF767/bsp/config.
-- Applying application-specific configuration from /home/rempler/repos/eclipse/kiso/examples/NucleoF767/c-leds/config.
-- Configuration headers merged in /home/rempler/repos/eclipse/kiso/build/config.
-- Performing Test has_no_tautological_compare
-- Performing Test has_no_tautological_compare - Success
-- Performing Test has_no_attributes
-- Performing Test has_no_attributes - Success
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
CMake Warning at docs/CMakeLists.txt:15 (message):
  Doxygen is needed to build the documentation.

-- Configuring incomplete, errors occurred!
See also "/home/rempler/repos/eclipse/kiso/build/CMakeFiles/CMakeOutput.log".
See also "/home/rempler/repos/eclipse/kiso/build/CMakeFiles/CMakeError.log".
rempler@RT-Z0ITK:~/r/e/kiso:master$
```

HansUweRempler, 2020-03-13:
```
rempler@RT-Z0ITK:~$ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.3 LTS"
rempler@RT-Z0ITK:~$
```
```
rempler@RT-Z0ITK:~$ cmake --version
cmake version 3.10.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
rempler@RT-Z0ITK:~$
```

elbosch, 2020-03-13:
@HansUweRempler what is the output on your machine if you execute the following:
wslpath -w /home/rempler/repos/eclipse/kiso/build/applications/c-leds/application.hex

HansUweRempler, 2020-03-16:
Okay, I tried the following but it looks like the binary wasn't created.
```
rempler@RT-Z0ITK:~/r/e/kiso:master$ git fetch
remote: Enumerating objects: 18, done.
remote: Counting objects: 100% (18/18), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 27 (delta 10), reused 10 (delta 10), pack-reused 9
Unpacking objects: 100% (27/27), done.
From https://github.com/eclipse/kiso
   9c2ba6a..fc0fe82  master     -> origin/master
   94d5e14..8fc0a79  feature/#17-add-test-application-and-adapt-bsp-stm32f7 -> origin/feature/#17-add-test-application-and-adapt-bsp-stm32f7
rempler@RT-Z0ITK:~/r/e/kiso:master$ git pull
Updating 9c2ba6a..fc0fe82
Fast-forward
 examples/NucleoF767/c-leds/readme.md | 69 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++-
 1 file changed, 68 insertions(+), 1 deletion(-)
rempler@RT-Z0ITK:~/r/e/kiso:master$ cmake . -Bbuild -DKISO_BOARD_NAME=NucleoF767
-- Looking for sysroot: /usr/lib/arm-none-eabi
------------- KISO CONFIG -------------
Building Kiso tests:   OFF
  ... with coverage:   ON
Kiso Board Path:       boards/NucleoF767
Kiso OS:               freertos
Kiso Application Path: examples/NucleoF767/c-leds
Project Config Path:   /home/rempler/repos/eclipse/kiso/config
------------- KISO CONFIG -------------
-- JLink not found in PATH. "flash" target unavailable!
ERROR JLink not found in path!
/bin/wslpath: /home/rempler/repos/eclipse/kiso/build/applications/c-leds/application.hex: Invalid argument
CMake Error at cmake/FlashTarget.cmake:65 (string):
  string sub-command STRIP requires two arguments.
Call Stack (most recent call first):
  examples/NucleoF767/c-leds/CMakeLists.txt:22 (CREATE_FLASH_TARGET_JLINK)

/bin/wslpath: /home/rempler/repos/eclipse/kiso/build/applications/c-leds/flash.jlink: Invalid argument
CMake Error at cmake/FlashTarget.cmake:83 (string):
  string sub-command STRIP requires two arguments.
Call Stack (most recent call first):
  examples/NucleoF767/c-leds/CMakeLists.txt:22 (CREATE_FLASH_TARGET_JLINK)

-- Applying board-specific configuration from /home/rempler/repos/eclipse/kiso/boards/NucleoF767/bsp/config.
-- Applying application-specific configuration from /home/rempler/repos/eclipse/kiso/examples/NucleoF767/c-leds/config.
-- Configuration headers merged in /home/rempler/repos/eclipse/kiso/build/config.
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
CMake Warning at docs/CMakeLists.txt:15 (message):
  Doxygen is needed to build the documentation.

-- Configuring incomplete, errors occurred!
See also "/home/rempler/repos/eclipse/kiso/build/CMakeFiles/CMakeOutput.log".
See also "/home/rempler/repos/eclipse/kiso/build/CMakeFiles/CMakeError.log".
rempler@RT-Z0ITK:~/r/e/kiso:master$ wslpath -w /home/rempler/repos/eclipse/kiso/build/applications/c-leds/application.hex
wslpath: /home/rempler/repos/eclipse/kiso/build/applications/c-leds/application.hex: Invalid argument
```

```
rempler@RT-Z0ITK:~/r/e/kiso:master$ ls -l build/applications/c-leds/
total 0
drwxrwxrwx 1 rempler rempler 4096 Mar 13 10:11 CMakeFiles/
-rw-rw-rw- 1 rempler rempler  154 Mar 16 10:43 flash.jlink
```

HansUweRempler, 2020-03-23:
@elbosch I have the felling that there's an issue with my cross compiler toolchain (version?). I think Saja and myself followed the quick start very and ended up with the same setup. Could it be that you set up your Linux/WSL/... differently to the http://kiso.rempler.de:1313/user-guide/quick_start.html ?

Look at the error log from above. IMHO this isn't related to JLink, Doxygen or whatsoever. 

```
rempler@RT-Z0ITK:~/r/e/kiso:master$ cat /home/rempler/repos/eclipse/kiso/build/CMakeFiles/CMakeError.log
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /usr/bin/arm-none-eabi-gcc
Build flags:
Id flags:

The output was:
1
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): In function `exit':
/build/newlib-jo3xW1/newlib-2.4.0.20160527/build/arm-none-eabi/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/exit.c:70: undefined reference to `_exit'
collect2: error: ld returned 1 exit status
```

```
rempler@RT-Z0ITK:~/r/e/kiso:master$ sudo apt install gcc-arm-none-eabi
[sudo] password for rempler:
Reading package lists... Done
Building dependency tree
Reading state information... Done
gcc-arm-none-eabi is already the newest version (15:6.3.1+svn253039-1build1).
0 upgraded, 0 newly installed, 0 to remove and 25 not upgraded.
```

AleksandraNedelkova, 2020-03-24:
Yesterday I tried the following and it worked I was able to build the project. I've added quotes at two lines in cmake/FlashTarget.cmake file(line 65 and 83):

if(WSLPATH)
        # 'wslpath -w' to be read directly inside jlink
        execute_process(COMMAND ${WSLPATH} -w ${HEX_TARGET_PATH} OUTPUT_VARIABLE WIN_TARGET_PATH)
line 65:        string(STRIP "${WIN_TARGET_PATH}" HEX_TARGET_PATH)
    endif()

    if(WSLPATH)
        # 'wslpath -m' so it's not escaped improperly
        execute_process(COMMAND ${WSLPATH} -m ${SCRIPT_PATH} OUTPUT_VARIABLE WIN_SCRIPT_PATH)
 line 83:       string(STRIP "${WIN_SCRIPT_PATH}" SCRIPT_PATH)
    endif()

elbosch, 2020-03-24:
Hi Aleks, thank you for your hint. Looks like this should fix it.. did you have spaces in the path?

AleksandraNedelkova, 2020-03-24:
There are no spaces between quotes.

BoykoKazakov, 2020-03-24:
The problem here is how you treat strip with empty string

elbosch, 2020-03-24:
cool. thanks.

HansUweRempler, 2020-03-24:
Yes, the @AleksandraNedelkova fix worked for me - thank you!
```
string(STRIP "${WIN_TARGET_PATH}" HEX_TARGET_PATH)
string(STRIP "${WIN_SCRIPT_PATH}" SCRIPT_PATH)
```

HansUweRempler, 2020-03-27:
Fixed

closed 2020-03-27 by HansUweRempler